dirtyjtag/DirtyJTAG #8: Create procedure for ST-Link firmware backup and restoration

jeanthom, 2017-12-03:
It would be nice to backup the ST-Link original firmware and put it back if needed. Maybe this could be done using [stlink](https://github.com/texane/stlink).

jeanthom, 2018-01-02:
This doesn't seems to be possible : https://lujji.github.io/blog/reverse-engineering-stlink-firmware-part2/
Removing read lock erases the whole flash content...

zoobab, 2018-01-03:
It would be good to at least put a WARNING: YOU CANNOT GO BACK AND REFLASH THE ORIGINAL FIRMWARE so that users know it is a one way trip.

andreygursky, 2018-01-03:
Or better: give a hint how to query the current firmware version. And a link to the available firmwares from ST.

Since a little while one is allowed to flash special version of firmware from SEGGER J-Link. Hopefully, it is not really a one-way ticket. Or maybe it is made with some proprietary info.

jeanthom, 2018-01-03:
It is possible to dump the bootloader : https://github.com/lujji/st-link-dumper
But the process is tedious

jeanthom, 2018-01-23:
I am currently in the process of allowing ST-Link dongles to use DirtyJTAG without erasing the original bootloader. This remove the need for ST-Link firmware backup (as you can restore it using ST's firmware update utility).

closed 2018-01-23 by jeanthom