dirtyjtag/DirtyJTAG #9: Automated JTAG test procedure

jeanthom, 2017-12-03:
An automated JTAG test procedure using an STM32 dev boards could be used for regression tests.

zoobab, 2017-12-13:
Trying now with another bluepill board as target device...

zoobab, 2017-12-21:
I have documented how to add another bluepill as JTAG target.

However, adding more manufacturers is still needed to detect "ARM Ltd" and "Thomson/SGS", they are not in the MANUACTURERS file.

zoobab, 2018-03-29:
I have tried to connect the bluepill to its JTAG pins, but the "detect" in urjtag gives a "TDO stuck at 1".

jeanthom, 2020-01-22:
I've been playing with a spare CPLD board lately to test my JTAG connection. It helped me to check whether basic JTAG pins were working.

I'd like to build a JTAG test jig around LambdaConcept's [jtagtap](https://github.com/lambdaconcept/jtagtap) and Gowin's cheap FPGAs that are supported by [project Apicula](https://github.com/pepijndevos/apicula). With a [LiteScope](https://github.com/enjoy-digital/litescope) onboard that would be a pretty rad test setup!