dirtyjtag/DirtyJTAG #14: DirtyJTAG install using DFU

jeanthom, 2017-12-05:
ST-Link v2 clone seems to have a DFU bootloader that allow for encrypted firmware upgrades. This will require a specific linker script (trivial) and bootloader patching (in the current situation, the DFU bootloader would not exit DFU until we would launch an st-info command).

Interesting links :
 * [https://lujji.github.io/blog/installing-blackmagic-via-stlink-bootloader/](https://lujji.github.io/blog/installing-blackmagic-via-stlink-bootloader/)
 * [https://github.com/lujji/st-decrypt](https://github.com/lujji/st-decrypt)

zoobab, 2017-12-13:
I could flash it with a usb-serial dongle, and powering it either via 5V or 3.3V.

Before flashing, you have to put the BOOT0 to 1, and keep BOOT1 at 0.

Then launch stm32flash:

```
$ stm32flash -w dirtyjtag.bluepill.bin -v -g 0x8000000 /dev/ttyUSB0
stm32flash 0.5

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Interface serial_posix: 57600 8E1
Version      : 0x22
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0410 (STM32F10xxx Medium-density)
- RAM        : 20KiB  (512b reserved by bootloader)
- Flash      : 128KiB (size first sector: 4x1024)
- Option RAM : 16b
- System RAM : 2KiB
Write to memory
Erasing memory
Wrote and verified address 0x08002028 (100.00%) Done.

Starting execution at address 0x08000000... done.
```

Then unplug it, put back the BOOT0 jumper to 0, and plug it in USB, you should see the "1209:c0ca InterBiometrics" in lsusb.

zoobab, 2017-12-13:
Will make a pull request with some pictures of the setup.

zoobab, 2017-12-13:
Ha sorry, this issue is about flashing over USB with DFU, not flashing with a serial dongle...

jeanthom, 2018-01-20:
https://github.com/jeanthom/stlink-tool solves the DFU uploading problem. The last thing to solve is #10 !

closed 2018-01-20 by jeanthom