dirtyjtag/DirtyJTAG #20: Add support for OpenOCD

zoobab, 2018-01-03:
Add support for OpenOCD.

jeanthom, 2020-03-09:
Work happening here: https://github.com/jeanthom/openocd-dirtyjtag

FFY00, 2021-01-29:
Hi, sorry to bother! How is this going? Is it still a WIP/unstable, or something that could be possibly upstreamed?

jeanthom, 2021-01-29:
Hi, no worries! :) Unfortunately it is currently in an unstable state... @brainstorm had success with an ESP32-S2 but I'm still struggling to get it to work with ECP5 (https://github.com/jeanthom/openocd-dirtyjtag/issues/2)

FFY00, 2021-01-29:
Okay, thanks, I will keep an eye out. If I have time, I will try to help out, but I doubt I will for the next few months :confused:

brainstorm, 2021-01-30:
Fast moving target, see https://github.com/jeanthom/openocd-dirtyjtag/issues/3 ;)

jeanthom, 2022-12-02:
Upstream in progress: https://review.openocd.org/c/openocd/+/7344

jonas-s-s-s, 2024-10-12:
> Upstream in progress: https://review.openocd.org/c/openocd/+/7344

No update on this in years? Still no way to use this with openOCD?

zoobab, 2024-10-12:
In the meantime I will try to make a static build of this OpenOCD branch.

jeanthom, 2024-10-12:
@jonas-s-s-s Unfortunately no progress. Getting things into mainline OpenOCD is not that easy (the codebase is not that easy to get in, high expectations of the maintainers) which explains why there are tons of vendor forks (eg. ESP32, Raspberry Pi, RISC-V, STMicroelectronics, Infineon, Texas Instruments, etc.).

jonas-s-s-s, 2024-10-12:
If this project had working OpenOCD then as far as I'm aware it'd be the only true JTAG debugger firmware compatible with RP2040.

My plan was to use the RP2040 to debug an RPi Zero, as it uses an older chip which doesn't support Serial Wire Debug. However I already gave up as it seems that using Qemu will be better than trying the hardware debugger approach.

zoobab, 2024-10-13:
"If this project had working OpenOCD"

We have a working OpenOCD, you just have to compile it yourself :-)

zoobab, 2025-11-25:
"I tried compiling that but it never worked."

What was your error?

Btw i am working on some openocd binaries.