dirtyjtag/DirtyJTAG #29: Add USB-serial capabilities

jeanthom, 2018-01-29:
STM32 dongles have 3 built-in serial IF, we could use them in addition to JTAG (bluepill targets mainly).

Inspiration : https://github.com/satoshinm/pill_serial

jeanthom, 2019-07-18:
We can't both use the three USART and keep the current Bluepill pinout.

Should we use only two out of the three USART or break the existing pinout instead?

zoobab, 2020-03-03:
@jeanthom sure break the pinout

jeanthom, 2020-03-03:
I'm not sure about breaking compatibility, maybe having a BluePill ALT (breaking compatibility with the current BluePill pinout) could be a solution. Adding a new platform/pinout is fairly easy.

zoobab, 2020-03-03:
Maybe reuse the usb-cdc code from BMP:

https://github.com/blacksphere/blackmagic/blob/master/src/platforms/common/cdcacm.c

zoobab, 2020-03-04:
Actually the code of pill_serial is from BMP:

https://github.com/satoshinm/pill_serial/blob/master/src/cdcacm.c

zoobab, 2020-03-09:
But since we are using unicoremx, the includes might be different...

jeanthom, 2020-03-09:
I think I might start by porting the pill_serial code to ucmx than integrate it to DirtyJTAG. This way I'll better understand the code, and avoid writing unnecessary code.

KnorkFox, 2023-12-25:
One serial port would be enough, on the blue pill board.