dirtyjtag/DirtyJTAG #56: Problem: Dockerfile does not build on docker hub

zoobab, 2019-11-08:
Problem: Dockerfile does not build on docker hub, seems docker hub does not like submodules:

https://hub.docker.com/repository/registry-1.docker.io/zoobab/dirtyjtag/builds/3ded2054-f2f9-4d00-8d58-94c7809ddfbe

```
Step 16/22 : RUN git submodule sync
---> Running in 26afeae2b5f3
Synchronizing submodule url for 'unicore-mx'
fatal: Not a git repository: /src/bb7tgzbayspbeirfmajq3t4/.git/modules/unicore-mx
fatal: Not a git repository: /src/bb7tgzbayspbeirfmajq3t4/.git/modules/unicore-mx
fatal: Not a git repository: /src/bb7tgzbayspbeirfmajq3t4/.git/modules/unicore-mx
Removing intermediate container 26afeae2b5f3
---> d2c5456aa385
Step 17/22 : RUN git submodule update
---> Running in e85bb4aabfae
fatal: Not a git repository: /src/bb7tgzbayspbeirfmajq3t4/.git/modules/unicore-mx
Unable to find current revision in submodule path 'unicore-mx'
Removing intermediate container e85bb4aabfae
The command '/bin/sh -c git submodule update' returned a non-zero code: 1
```

jeanthom, 2020-01-24:
Did https://github.com/jeanthom/DirtyJTAG/commit/c4201c3c2064c2726533d8d5f852cb0872775f28 solve the issue?

zoobab, 2020-01-24:
No I have not found a solution yet...

jeanthom, 2022-11-28:
Obsoleted by #97

closed 2022-11-28 by jeanthom