dirtyjtag/DirtyJTAG #67: Can't build

FFY00, 2020-07-31:
```
$ make PLATFORM=stlinkv2
make -C /home/anubis/git/DirtyJTAG/unicore-mx lib/stm32/f1
make[1]: Entering directory '/home/anubis/git/DirtyJTAG/unicore-mx'
  GENHDR  include/unicore-mx/efm32/g/irq.json
  GENHDR  include/unicore-mx/efm32/gg/irq.json
  GENHDR  include/unicore-mx/efm32/lg/irq.json
  GENHDR  include/unicore-mx/efm32/tg/irq.json
  GENHDR  include/unicore-mx/lm3s/irq.json
  GENHDR  include/unicore-mx/lpc13xx/irq.json
  GENHDR  include/unicore-mx/lpc17xx/irq.json
  GENHDR  include/unicore-mx/lpc43xx/m0/irq.json
  GENHDR  include/unicore-mx/lpc43xx/m4/irq.json
  GENHDR  include/unicore-mx/nrf/51/irq.json
  GENHDR  include/unicore-mx/sam/3a/irq.json
  GENHDR  include/unicore-mx/sam/3n/irq.json
  GENHDR  include/unicore-mx/sam/3s/irq.json
  GENHDR  include/unicore-mx/sam/3u/irq.json
  GENHDR  include/unicore-mx/sam/3x/irq.json
  GENHDR  include/unicore-mx/sam/4l/irq.json
  GENHDR  include/unicore-mx/stm32/f0/irq.json
  GENHDR  include/unicore-mx/stm32/f1/irq.json
  GENHDR  include/unicore-mx/stm32/f2/irq.json
  GENHDR  include/unicore-mx/stm32/f3/irq.json
  GENHDR  include/unicore-mx/stm32/f4/irq.json
  GENHDR  include/unicore-mx/stm32/f7/irq.json
  GENHDR  include/unicore-mx/stm32/l0/irq.json
  GENHDR  include/unicore-mx/stm32/l1/irq.json
  GENHDR  include/unicore-mx/stm32/l4/irq.json
  GENHDR  include/unicore-mx/vf6xx/irq.json
  GENUCH  include/unicore-mx/common/dwc_otg.h
./scripts/uc-def/uc-def:240: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if inp[0] is not 'r':
  GENUCH  include/unicore-mx/stm32/common/st_usbfs_common.h
./scripts/uc-def/uc-def:240: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if inp[0] is not 'r':
  GENUCH  include/unicore-mx/stm32/common/st_usbfs_v1.h
./scripts/uc-def/uc-def:240: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if inp[0] is not 'r':
  GENUCH  include/unicore-mx/stm32/common/st_usbfs_v2.h
./scripts/uc-def/uc-def:240: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if inp[0] is not 'r':
  GENUCH  include/unicore-mx/stm32/f3/opamp.h
./scripts/uc-def/uc-def:240: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if inp[0] is not 'r':
  BUILD   lib/stm32/f1
/bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
/bin/sh: -c: line 1: syntax error: unexpected end of file
make[1]: *** [Makefile:70: lib/stm32/f1] Error 1
make[1]: Leaving directory '/home/anubis/git/DirtyJTAG/unicore-mx'
make: *** [Makefile:54: ucmx] Error 2
```

jeanthom, 2020-08-01:
Hi!

Thanks for your interest in DirtyJTAG. Your error seems to be related to [unicore-mx](https://gitlab.com/insane-adding-machines/unicore-mx). I'm using an old version in DirtyJTAG (haven't got the time lately to update it), could you try the newer version from their GitLab repository?

zoobab, 2020-08-01:
./build.sh is working fine here.

Otherwise the git submodule points to:

$ git submodule status
-a2a06059bdce11fb24d872e8d93e4455545ad65b unicore-mx

jeanthom, 2020-08-01:
@FFY00 what shell are you using?

closed 2020-08-11 by jeanthom