dirtyjtag/DirtyJTAG #71: superfluous -g 0x8000000 for stm32flash

nerdralph, 2020-12-06:
Since the boot0 jumper will be removed and the power cycled after flashing, there is no point in adding the "-g 0x8000000" to tell the bootloader to jump to the start of flash after writing.

jeanthom, 2020-12-07:
Hi,

This flag might be useful for debugging purposes? Actually I've never used stm32flash. @zoobab what's your opinion on this?

zoobab, 2020-12-07:
He is probably right, this can be removed.

Before discovering stm32flash, I was using another python tool which I had real problems with:

http://www.zoobab.com/bluepill-arduinoide

jeanthom, 2020-12-07:
Ok I'll remove it from the docs then. Thanks @nerdralph for raising this issue, and thanks @zoobab for your input.

commit 7fa7fd09 (2020-12-07): Remove "-g 0x8000000" argument in stm32flash command example

Fixes #71
closed 2020-12-07 by jeanthom