dirtyjtag/DirtyJTAG #81: Port to stm32f072

madushan1000, 2021-01-17:
Most of the jlink ob clones on [aliexpress](https://de.aliexpress.com/item/4000249803796.html?spm=a2g0o.productlist.0.0.287c27c6YYLOPS&algo_pvid=51010f69-66c5-4933-badd-efdef4829b04&algo_expid=51010f69-66c5-4933-badd-efdef4829b04-0&btsid=2100bddf16109220535694671e960f&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) comes with this chip. As far as I could see, these clones have 4 3V tolerant pins, and 2 5V tolerant pins for VCOM. There also seems to be a hardwired reset pin too. 
I think this document has the correct pinmap(https://github.com/idrnyu/dllGetSt072JlinkOb/blob/master/%E7%94%B5%E8%B7%AF072%E6%B3%A8%E6%84%8F.txt)

I'm wondering if it would be possible to port dirtyjtag to this chip, it supports usb dfu too.

zoobab, 2021-01-18:
Jlink has many hardware revisions (v9, v10.4, etc...), do you have any documentation (photo of the PCB?) on which hardware revision has this stm32f072?

I think I also have this Nucleo board in my stock: https://www.st.com/en/evaluation-tools/nucleo-f072rb.html

madushan1000, 2021-01-18:
The chip is used in jlink "on board" clones, It's a simple jlink firmware with only swd + vcom support. I don't think it has any particular hardware revision.
1) https://www.segger.com/products/debug-probes/j-link/models/j-link-ob/
2) https://www.segger.com/downloads/jlink/UM08024_JLinkOBSTM32F072.pdf

I think [this](https://github.com/x893/CMSIS-DAP/tree/master/HW_STM32) is at least a partial schematic(header pinout is a little different). 

I'll get a photo of the pcb when I have some time.

madushan1000, 2021-01-18:
Also, the jlink firmware requires that the pinout to be a certain way. so most of these clones have the bellow pinout. So I don't think the proper schematic is strictly required.

```
PA0 ----- 1K -----|<|---+ 3.3V   Activity LED 
PA1 ----- 100 om ----TARGET RESET 
PA2 ----- 100 om ---- TARGET SWCLK 
PA3 -------------------- TARGET SWO 
PA4 ----- 100 om ---- TARGET SWDIO 

PA9 ----- 100 om ---- (COM TX) TARGET RX 
PA10 --- 100 om ---- (COM RX) TARGET TX 
```

jeanthom, 2021-01-18:
I wouldn't bother porting to J-Link hardware clones, it's so hard to purchase a specific hardware revision with the right MCU... But otherwise porting DirtyJTAG to STM32F0 should be feasible. You'd have to edit the Makefile to enable STM32F0 builds for unicore-mx, change at least two compiler flags (-DSTM32F1 and -mcpu), and edit the frequency definition macro.

The only thing I own that has an STM32F072 is my soldering iron, and I'm not sure that it would be useful at all to port DirtyJTAG on it :laughing:

madushan1000, 2021-01-18:
If it would be that easy, I can give it a try myself. The only problem is getting the clone into dfu mode. the clearance between pins are so tiny I'm not sure if I'd short something. I'll get you some pictures in a couple of hours.

madushan1000, 2021-01-19:
Looks like no level shifters :/ 
[image]
[image]
[image]

jeanthom, 2021-01-19:
Ah that's sad... You might be able to reprogram these using the P1 connector.

zoobab, 2021-01-19:
You could try to flash via the P1 connector, if you don't have an stlinkv2, maybe use a usb-serial adaptor like documented.

madushan1000, 2021-01-19:
These chips has usb dfu, I'm wondering if I can get that to work. I don't have a jtag/swd adapter handy at the moment.

jeanthom, 2021-04-03:
I'm closing this issue because I'm not interested in STM32F0 support at the moment. Feel free to reopen this issue if you are willing to take the challenge ;)

closed 2021-04-03 by jeanthom