dirtyjtag/DirtyJTAG #82: Alpine as build distro?

zoobab, 2021-01-19:
Hi,

I just found out that someone packages the cross-compilers for ARM for Alpine, so I rewrote the Dockerfile to use it:

```
FROM alpine:3.12
MAINTAINER Benjamin Henrion <zoobab@gmail.com>
LABEL Description="DirtyJTAG firmware for STM32 Bluepill board"
RUN echo -e "http://dl-cdn.alpinelinux.org/alpine/edge/main\nhttp://dl-cdn.alpinelinux.org/alpine/e
dge/testing\nhttp://dl-cdn.alpinelinux.org/alpine/edge/community" >> /etc/apk/repositories
RUN apk update
RUN apk add gcc-arm-none-eabi sudo make python2 git bash
RUN mkdir -pv /code
COPY . /code
WORKDIR /code
RUN git submodule init
RUN git submodule sync
RUN git submodule update
RUN make PLATFORM=bluepill
RUN make PLATFORM=stlinkv2
RUN make PLATFORM=stlinkv2dfu
RUN make PLATFORM=baite
RUN make PLATFORM=olimexstm32h103
RUN make PLATFORM=stlinkv2white
```
But when I build it I have this weird shell error, adding bash does not help: 

```
 ---> Running in 4f4ec574bb98
make -C /code/unicore-mx lib/stm32/f1
make[1]: Entering directory '/code/unicore-mx'
  GENHDR  include/unicore-mx/stm32/f3/irq.json
  GENHDR  include/unicore-mx/stm32/f1/irq.json
  GENHDR  include/unicore-mx/stm32/l1/irq.json
  GENHDR  include/unicore-mx/stm32/f7/irq.json
  GENHDR  include/unicore-mx/stm32/f4/irq.json
  GENHDR  include/unicore-mx/stm32/l0/irq.json
  GENHDR  include/unicore-mx/stm32/f0/irq.json
  GENHDR  include/unicore-mx/stm32/l4/irq.json
  GENHDR  include/unicore-mx/stm32/f2/irq.json
  GENHDR  include/unicore-mx/lm3s/irq.json
  GENHDR  include/unicore-mx/nrf/51/irq.json
  GENHDR  include/unicore-mx/sam/3x/irq.json
  GENHDR  include/unicore-mx/sam/3s/irq.json
  GENHDR  include/unicore-mx/sam/3n/irq.json
  GENHDR  include/unicore-mx/sam/3a/irq.json
  GENHDR  include/unicore-mx/sam/3u/irq.json
  GENHDR  include/unicore-mx/sam/4l/irq.json
  GENHDR  include/unicore-mx/lpc13xx/irq.json
  GENHDR  include/unicore-mx/lpc43xx/m0/irq.json
  BUILD   lib/stm32/f1
/bin/sh: syntax error: unterminated quoted string
make[1]: *** [Makefile:70: lib/stm32/f1] Error 2
make[1]: Leaving directory '/code/unicore-mx'
make: *** [Makefile:54: ucmx] Error 2
The command '/bin/sh -c make PLATFORM=bluepill' returned a non-zero code: 2
```

Any idea?

zoobab, 2021-01-19:
This patch to the Makefile spacings fixed the issue:

https://gitlab.com/encrypt/unicore-mx/-/commit/ae26cf298ab9508082c62ecc9aa9fe02330924fb

jeanthom, 2021-01-19:
Gotta update unicore-mx sometime soon (#66), is this patch present in the official upstream?

zoobab, 2021-01-19:
Yes it is in the upstream.

zoobab, 2021-01-19:
Tried upstream, it seems to build, will try to flash it on a bluepill...

zoobab, 2021-12-05:
Now Alpine 3.14 and 3.15 have the ARM cross-compilers (finally!), you can use:

FROM alpine:3.15
RUN apk add python2 make gcc-arm-none-eabi newlib-arm-none-eabi git

zoobab, 2021-12-06:
This works for me:

FROM alpine:3.15
RUN apk add python2 make gcc-arm-none-eabi newlib-arm-none-eabi git
RUN mkdir -pv /code
COPY . /code
WORKDIR /code
RUN git submodule init
RUN git submodule sync
RUN git submodule update
RUN make PLATFORM=bluepill
RUN make PLATFORM=stlinkv2
RUN make PLATFORM=stlinkv2dfu
RUN make PLATFORM=baite
RUN make PLATFORM=olimexstm32h103
RUN make PLATFORM=stlinkv2white

jeanthom, 2022-11-28:
Fixed in #97

closed 2022-11-28 by jeanthom