dirtyjtag/DirtyJTAG #87: Add some benchmarks table

zoobab, 2021-03-16:
Add some benchmarks table, comparing the speed with other dongles.

With a versaloon stm32 and openocd, flashing a file can take a minute, while it takes 1s with an FT2232H.

jeanthom, 2021-04-03:
What kind of target should use for the benchmark?

 * FPGA? I know ECP5 would be highly appreciated
 * Micro-controllers?

zoobab, 2022-11-16:
Target: another stm32f103 bluepill board.

This discussion about performances was raised by some openocd maintainers, that would be good to have some numbers.

KnorkFox, 2023-12-18:
Since you said, that dirtyjtag does not try to be fast, it is just dirty and works, is performance still an issue here?

If yes, i could imagine reading a flash rom, as a proper benchmark. I intend to use dirtyjtag as a proprammer with flashrom.

I have two boards bluepill, and one board raspberry pico.

KnorkFox, 2023-12-25:
According to pico-serprog: 
"This takes about 17 seconds to read the 8MiB BIOS chip of an X200."

A blue pill should be faster.

jeanthom, 2023-12-25:
Hi @KnorkFox,

To give you the context, I started DirtyJTAG with the "I just want it to work, no matter how slow it will be" mindset, hence the "dirty" in DirtyJTAG. The project has since received significant contributions from @phdussud (and other contributiors IIRC) which have greatly improved performance (JTAG bitbang -> faster JTAG bitbang -> SPI whenever possible).

While I'm not concerned about performance issues, it is still useful for users to be able to assess the performance of a JTAG adapter. When iterating on FPGA projects you don't want to spend 5 minutes on each bitstream flash.

I haven't got the opportunity to find the motivation to work on this project, which is why this issue is still open, but I'd love to see someone tackling this issue!

KnorkFox, 2023-12-25:
Me too. As you might have guessed, this is way beyond my possibilitys. We have to wait for someone else.

phdussud, 2023-12-26:
To make progress, we need to pinpoint to a real scenario with a documented process:
Description of the device, description of the software used for the programming, and timing comparison using different adapters using the same software. For your information, I used OpenFPGALoader and an ECP5 FPGA board when I tuned the program. This combination of device/software hits the limit of a USB Full speed link(12Mbs).  FT2232H used High Speed with is 480Mbs so it can be faster. This brings another perspective to this effort. Nowadays you can pick up an Adafruit FT232H Breakout for $15. I believe it is compatible with the FT2232H software. Is it worthwhile to put a lot more effort into DirtyJtag to make incremental performance progress when you can spend $10 more and be there?

KnorkFox, 2023-12-27:
Speaking about speed limits. For my usage, 12 Mb per second would be enogh to read or write a memory chip occationally. I dont really need super high speed.

phdussud, 2023-12-27:
I think you get close to this if you use OpenFPGALoader to write the memory. I don't know about other tools. It turns out that the sequence of DirtyJtag instructions is important to get good performance.