dirtyjtag/DirtyJTAG #90: Required flash size for STM32F1

xiak95, 2021-08-15:
Hi, 

I would like to buy bluepill to flash it with DirtyJTAG, and I wonder what is the required flash size for the firmware ! 

There appear to be two bluepill versions on aliexpress, one with STM32F103C6T6 and the other with STM32F103C8T6 ! 

Would dirtyjtag fit on the earlier, or do I need the later ? 

Thanks !

jeanthom, 2021-08-17:
Hi,

DirtyJTAG being very small (<10kB), you can use both versions. However be aware that counterfeit STM32F103 are becoming more and more prevalent on AliExpress!

drnedm, 2021-08-22:
I just received a STM32F103C6T6 based blue pill, and it doesn't enumerate as a USB device after flashing DirtyJTAG. 
It works nicely as an STM32DUINO, including serial via USB, so the hardware seems fine in principle. Unsure how to debug this. I tried both the released binary as well as compiling it myself, same result. Should this go into a new issue?

jeanthom, 2021-08-24:
Hi @drnedm, I never came across F103C6x devices before. Could you please file an issue? Also where did you get a board with this variant of the F103?

closed 2021-08-24 by jeanthom