dirtyjtag/DirtyJTAG #92: Build fails on Debian 11

richardklingler, 2021-11-23:
A frsh recursive checkout as documented fails on building it:

```
me@blender:~/develop/dirtyjtag$ make PLATFORM=bluepill
make -C /home/me/develop/dirtyjtag/unicore-mx lib/stm32/f1
make[1]: Entering directory '/home/me/develop/dirtyjtag/unicore-mx'
  GENHDR  include/unicore-mx/efm32/tg/irq.json
  GENHDR  include/unicore-mx/efm32/g/irq.json
  GENHDR  include/unicore-mx/efm32/gg/irq.json
  GENHDR  include/unicore-mx/efm32/lg/irq.json
  GENHDR  include/unicore-mx/stm32/f1/irq.json
  GENHDR  include/unicore-mx/stm32/f3/irq.json
  GENHDR  include/unicore-mx/stm32/f2/irq.json
  GENHDR  include/unicore-mx/stm32/f0/irq.json
  GENHDR  include/unicore-mx/stm32/f4/irq.json
  GENHDR  include/unicore-mx/stm32/l1/irq.json
  GENHDR  include/unicore-mx/stm32/f7/irq.json
  GENHDR  include/unicore-mx/stm32/l0/irq.json
  GENHDR  include/unicore-mx/stm32/l4/irq.json
  GENHDR  include/unicore-mx/lm3s/irq.json
  GENHDR  include/unicore-mx/vf6xx/irq.json
  GENHDR  include/unicore-mx/lpc43xx/m0/irq.json
  GENHDR  include/unicore-mx/lpc43xx/m4/irq.json
  GENHDR  include/unicore-mx/nrf/51/irq.json
  GENHDR  include/unicore-mx/lpc13xx/irq.json
  GENHDR  include/unicore-mx/lpc17xx/irq.json
  GENHDR  include/unicore-mx/sam/3x/irq.json
  GENHDR  include/unicore-mx/sam/3u/irq.json
  GENHDR  include/unicore-mx/sam/3a/irq.json
  GENHDR  include/unicore-mx/sam/4l/irq.json
  GENHDR  include/unicore-mx/sam/3s/irq.json
  GENHDR  include/unicore-mx/sam/3n/irq.json
  GENUCH  include/unicore-mx/stm32/f3/opamp.h
  GENUCH  include/unicore-mx/stm32/common/st_usbfs_v1.h
  GENUCH  include/unicore-mx/stm32/common/st_usbfs_common.h
  GENUCH  include/unicore-mx/stm32/common/st_usbfs_v2.h
  GENUCH  include/unicore-mx/common/dwc_otg.h
  BUILD   lib/stm32/f1
/bin/sh: 1: Syntax error: Unterminated quoted string
make[1]: *** [Makefile:70: lib/stm32/f1] Error 2
make[1]: Leaving directory '/home/me/develop/dirtyjtag/unicore-mx'
make: *** [Makefile:54: ucmx] Error 2

```

Furthermore I had to change line 240 in scripts/uc-def/uc-def from:

`	if inp[0] is not 'r':
`

to

`	if inp[0] != 'r':
`

zoobab, 2021-11-23:
Try this fix:

https://gitlab.com/insane-adding-machines/unicore-mx/-/commit/5ba1713e167dd9d9bb9e87d3659a106e39980b40

We have to move the submodule to gitlab, but I have not checked if it works with one of their latest version.

richardklingler, 2021-11-23:
Yes...that did it :-)

muchas gracias (o;

jeanthom, 2021-11-23:
Reopening the issue so that I don't forget to fix this :)

jeanthom, 2021-11-23:
Issue fixed in master (#66)

closed 2021-11-23 by jeanthom