dirtyjtag/DirtyJTAG #98: error: libusb error: libusb_init() failed: 6

dontdeadopeninside, 2022-10-21:
jtag> cable dirtyjtag
error: Couldn't connect to suitable USB device.
error: libusb error: libusb_init() failed: 6
jtag> 

Shows up properly in lsusb and dmesg

jeanthom, 2022-10-22:
Hi,

It probably is a permission error. The short and hacky solution would be to run openFPGALoader/UrJTAG/whatever as a root user. The long and cleaner solution is to setup udev rules.

dontdeadopeninside, 2022-10-22:
Unfortunately running as root and sudo both result in the same result, same with openFPGALoader

jeanthom, 2022-10-23:
Don't worry I can reopen issues.

Can you give me more information that could help me reproduce the issue?

 - Which hardware platform are you using?
 - What firmware version are you using? Did you compile it yourself?
 - What operating system are you using?

jeanthom, 2022-11-01:
Closing issue due to lack to reply from original poster. Feel free to comment if the issue persists, I'll re-open the issue.

closed 2022-11-01 by jeanthom