dirtyjtag/DirtyJTAG #106: STM32F103C8T6 device descriptor error

metalstarlight, 2023-01-09:
I'm trying to load the firmware into the controller, in all cases (even in previous compiled versions) dmesg returns the following:

`[14276.055065] usb 1-2: new full-speed USB device number 51 using xhci_hcd
[14281.314934] usb 1-2: device descriptor read/all, error -110
[14281.442817] usb 1-2: new full-speed USB device number 52 using xhci_hcd
[14286.690929] usb 1-2: device descriptor read/all, error -110
[14286.691049] usb usb1-port2: attempt power cycle
[14287.342826] usb 1-2: new full-speed USB device number 53 using xhci_hcd
[14292.578949] usb 1-2: device descriptor read/8, error -110
[14297.958937] usb 1-2: device descriptor read/8, error -110
[14298.202826] usb 1-2: new full-speed USB device number 54 using xhci_hcd
[14303.334925] usb 1-2: device descriptor read/8, error -110
[14308.706942] usb 1-2: device descriptor read/8, error -110
[14308.814946] usb usb1-port2: unable to enumerate USB device
`

However for example Black Magic Probe firmware works correctly on the same device.
Is this "special" bugged bluepill? 
All hardware is OK (R10=1.5k), soldering is fine. i've tried with noloader parameter, 2k,4k. Same effect. Used to uploading ST-link v2.

jeanthom, 2023-01-16:
Hi Kim,

Your issue is very concerning. To help me debug your issue, could you please report/try the following things:

1. Report the git hash of DirtyJTAG if you've build it from source, or the release version of you were using the release builds.
2. Report all the parameters used for programming the STM32 (I'm assuming you are using stlink command line tools).
3. Have you tried replugging your Blue Pill after reprogramming? I haven't been able to get USB re-enumeration you work reliably on this platform.

Thanks,
Jean.

metalstarlight, 2023-01-17:
Thanks, Jead. Here is a log:
Building from the source:
`st-flash write dirtyjtag.bluepill.noloader.bin 0x8000000
st-flash 1.7.0
2023-01-17T02:00:32 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 128 KiB flash in at least 1 KiB pages.
file dirtyjtag.bluepill.noloader.bin md5 checksum: 3a94dd19ada294cd9b8399b599487f3, stlink checksum: 0x000dc45b
2023-01-17T02:00:32 INFO common.c: Attempting to write 9232 (0x2410) bytes to stm32 address: 134217728 (0x8000000)
2023-01-17T02:00:32 INFO common.c: Flash page at addr: 0x08000000 erased
2023-01-17T02:00:32 INFO common.c: Flash page at addr: 0x08000400 erased
2023-01-17T02:00:32 INFO common.c: Flash page at addr: 0x08000800 erased
2023-01-17T02:00:32 INFO common.c: Flash page at addr: 0x08000c00 erased
2023-01-17T02:00:32 INFO common.c: Flash page at addr: 0x08001000 erased
2023-01-17T02:00:32 INFO common.c: Flash page at addr: 0x08001400 erased
2023-01-17T02:00:32 INFO common.c: Flash page at addr: 0x08001800 erased
2023-01-17T02:00:32 INFO common.c: Flash page at addr: 0x08001c00 erased
2023-01-17T02:00:32 INFO common.c: Flash page at addr: 0x08002000 erased
2023-01-17T02:00:32 INFO common.c: Flash page at addr: 0x08002400 erased
2023-01-17T02:00:32 INFO common.c: Finished erasing 10 pages of 1024 (0x400) bytes
2023-01-17T02:00:32 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2023-01-17T02:00:32 INFO flash_loader.c: Successfully loaded flash loader in sram
2023-01-17T02:00:32 INFO flash_loader.c: Clear DFSR
 10/ 10 pages written
2023-01-17T02:00:32 INFO common.c: Starting verification of write complete
:~/DirtyJTAG/src$ md5sum dirtyjtag.bluepill.noloader.bin 
3a94dd19ad0a294cd9b8399b599487f3  dirtyjtag.bluepill.noloader.bin
`
So, there in some warnings during compilation process.

Using pre-compilated release [Version 2.0]([https://github.com/dirtyjtag/DirtyJTAG/releases/tag/2.0]):
`$ st-flash write dirtyjtag.bluepill.noloader.bin 0x8000000
st-flash 1.7.0
2023-01-17T02:13:37 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 128 KiB flash in at least 1 KiB pages.
file dirtyjtag.bluepill.noloader.bin md5 checksum: dc87643c89d1b379476ddda919fbc853, stlink checksum: 0x000db627
2023-01-17T02:13:37 INFO common.c: Attempting to write 9224 (0x2408) bytes to stm32 address: 134217728 (0x8000000)
2023-01-17T02:13:37 INFO common.c: Flash page at addr: 0x08000000 erased
2023-01-17T02:13:37 INFO common.c: Flash page at addr: 0x08000400 erased
2023-01-17T02:13:37 INFO common.c: Flash page at addr: 0x08000800 erased
2023-01-17T02:13:37 INFO common.c: Flash page at addr: 0x08000c00 erased
2023-01-17T02:13:37 INFO common.c: Flash page at addr: 0x08001000 erased
2023-01-17T02:13:37 INFO common.c: Flash page at addr: 0x08001400 erased
2023-01-17T02:13:37 INFO common.c: Flash page at addr: 0x08001800 erased
2023-01-17T02:13:37 INFO common.c: Flash page at addr: 0x08001c00 erased
2023-01-17T02:13:37 INFO common.c: Flash page at addr: 0x08002000 erased
2023-01-17T02:13:37 INFO common.c: Flash page at addr: 0x08002400 erased
2023-01-17T02:13:37 INFO common.c: Finished erasing 10 pages of 1024 (0x400) bytes
2023-01-17T02:13:37 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2023-01-17T02:13:37 INFO flash_loader.c: Successfully loaded flash loader in sram
2023-01-17T02:13:37 INFO flash_loader.c: Clear DFSR
2023-01-17T02:13:37 INFO flash_loader.c: Clear CFSR
2023-01-17T02:13:37 INFO flash_loader.c: Clear HFSR
 10/ 10 pages written
2023-01-17T02:13:37 INFO common.c: Starting verification of write complete
2023-01-17T02:13:37 INFO common.c: Flash written and verified! jolly good!`

Here is an output from ["Blue Pill Diagnostic"] (https://mecrisp-stellaris-folkdoc.sourceforge.io/bluepill-diagnostics-v1.6.html) tool:

`===========================================
Bluepill Diagnostics V1.640 - GPL3 Licensed
===========================================
h - test second Half of the 128KB flash declared for this chip
f - how much Flash is declared in the Flash Size Register ?
d - Print DBGMCU_IDCODE
a - STM32F103C8T6 Authenticity test, don't use with SWD/JTAG.
j - Jdec manufacturer id
e - Extra menu
q - Quit menu, enter the Forth command line
m - Main menu

Please wait, testing Flash ......

131072 flash is declared in the Flash size register at 0x1FFFF7E0

DBGMCU_IDCODE [@ 0xE0042000] = 0x20036410

STM32F103C8 authentication FAILED one or more tests:
----------------------------------------------------
FAIL - Declared flash not 65536
FAIL - DBGMCU_IDCODE is readable with no SWD/Jtag connected
PASS - Second 64KB flash block verified
PASS - JDEC manufacturer id IS STMicroelectronics

Jdec Continuation Code: 0x00
Jdec Identity Code: 0x20
JDEC manufacturer id: STMicroelectronics

----------
Extra Menu
----------
f - view First  64kb flash memory block: 0x00000 - 0x10000
s - view Second 64kb flash memory block: 0x10000 - 0x1FFFF
i - unique device Id
n - unique derived serial Number
a - fAQ
u - Usb faq
c - Credits
q - Quit back to main menu
m - Extra menu

UNIQUE DEVICE ID:
-----------------
BITS-95:64 | 0x41373931 | A791
BITS-63:32 | 0x31333915 | 139.
BITS-31:0  | 0x002D003F | .-.?

Unique Serial Number = 0x7029001B

`

jeanthom, 2023-01-17:
Thanks for your report. It appears that you have an undocumented counterfeit STM32. I found a user on the net facing similar issues with their STM32 board: https://github.com/qmk/qmk_firmware/issues/17605

Unfortunately I don't know where to investigate, and I can't justify spending a lot of time fixing buggy STM32 clones. If you ever find relevant resources about STM32 clones, or have a patch I could integrate, feel free to reach out.

closed 2023-01-17 by jeanthom