dirtyjtag/DirtyJTAG #108: TI TIVA TM4C JTAG

froloffw7, 2023-03-06:
Hello,
I'm trying to reflash TI MSP432P401R builtin XDS 110 JTAG programmer firmware, which was broken during the update and doesn't enter DFU mode.
TI suggest to use JTAG (seems this is the only possible way). As I don't have any TI XDS JTAG programmer, I'm trying to do the same with some other JTAG hardware.
I didn't achieve any success with Altera Byte Blaster neither with ST-Link v2, so I'm trying to do this with DirtyJtag+openocd(patched). It can detect the CPU, but then failed during initialization with error: 
Error: 103 214 bitq.c:308 bitq_execute_queue(): missing data from bitq interface.
Do you have any ideas what to check/change?
openocd configuration file, inspired by _board/ti_ek-tm4c1294xl.cfg_ 
```
#
# TM4C1294NCPDT Chip
#
# Author : 
# Version : 
#
source [find interface/dirtyjtag.cfg]
# transport select jtag

set WORKAREASIZE 0x8000
set CHIPNAME tm4c1294ncpdt
source [find target/stellaris.cfg]

```

Full openocd debug log attached.
Thank you!
[openocd.txt](https://github.com/dirtyjtag/DirtyJTAG/files/10899733/openocd.txt)

jeanthom, 2023-03-13:
Hi @froloffw7,

I haven't been very active on the OpenOCD driver. I had a brief look but I can't figure what's going wrong.

@phdussud do you have any advice or things they should try? Thanks :)

phdussud, 2023-03-13:
Hello, 
I looked at the log. I believe that the error comes from a lack of response from the TI chip, or a bug in the openocd driver or in the  dirtyjtag firmware. Are you using dirtyJtag V2?
This is going to be hard to debug. A similar debug session works with a STM2F4xx chip but that does not help you any. Sorry! 
The only thing I can suggest is to use a Logic Analyzer to spy on the Jtag interface and see if the exchange right before the error message makes sense. Alternatively, you can try an older version of the dirtyJtag support of openocd here
[https://sourceforge.net/u/phdussud/openocd/](https://sourceforge.net/u/phdussud/openocd/)